pal-robotics/tiago_robot #1: find speed_limit

olivier-stasse, 2016-09-22:
Hi Jordi,
Sorry for the stupid question but where is the package speed_limit specified in 
https://github.com/pal-robotics/tiago_robot/blob/indigo-devel/tiago_bringup/launch/tiago_bringup.launch#L26

Thanks in advance.
Best,
Olivier.

jordi-pages, 2016-09-22:
> indigo-devel/tiago_bringup/launch/tiago_bringup.launch#L26

Salut Olivier,

sorry for the convenience. We are uploading a new version of TIAGo
simulation and there are still some inconsistencies. I will let you know as
soon as this has been fixed.

Cheers

Jordi

-- 

Jordi Pages

TIAGo Product Manager

C/ Pujades 77-79, 4-4

08005 Barcelona, Spain

Skype jordi.pal-robotics

Tel +34 93 414 53 47

jordi.pages@pal-robotics.com

www.pal-robotics.com
Facebook <http://www.facebook.com/palrobotics1> | Twitter
<http://twitter.com/#%21/palrobotics> | YouTube
<http://www.youtube.com/user/PALRobotics> | Blog
<http://blog.pal-robotics.com/>

P Antes de imprimir este e-mail piense bien si es necesario hacerlo: El
medioambiente es cosa de todos.

AVISO DE CONFIDENCIALIDAD: Este mensaje y sus documentos adjuntos, pueden
contener información privilegiada y/o confidencial que está dirigida
exclusivamente a su destinatario.  Si usted recibe este mensaje y no es el
destinatario indicado, o el empleado encargado de su entrega a dicha
persona, por favor, notifíquelo inmediatamente y remita el mensaje original
a la dirección de correo electrónico indicada. Cualquier copia, uso o
distribución no autorizados de esta comunicación queda estrictamente
prohibida.

CONFIDENTIALITY NOTICE: This e-mail and the accompanying document(s) may
contain confidential information which is privileged and intended only for
the individual or entity to whom they are addressed.  If you are not the
intended recipient, you are hereby notified that any disclosure, copying,
distribution or use of this e-mail and/or accompanying document(s) is
strictly prohibited.  If you have received this e-mail in error, please
immediately notify the sender at the above e-mail address.

olivier-stasse, 2016-09-23:
Dear Jordi,
Thanks for your quick answer, I thus used the binaries packages and it worked.
Best,
Olivier.

closed 2016-09-23 by olivier-stasse

jordi-pages, 2020-02-19:
---------- Forwarded message ---------
From: Raul Suarez <raul.suarez@upc.edu>
Date: Tue, Feb 18, 2020 at 6:35 PM
Subject: tiago2 brazos
To: Jordi Pages <jordi.pages@pal-robotics.com>, Carlos Vivas <
carlos.vivas@pal-robotics.com>

Hola Jordi, hola Carlos,
tal como quedamos en el pick&pack, os escribo para pediros una oferta del
upgrade del Tiago que tenemos nosotros, de un brazo, al nuevo Tiago de dos.
Ya me diréis precio, condiciones, viabilidad....
Un cordial saludo
Raúl

-- 

*      Dr. Raúl Suárez Feijóo*
       Director de Recerca

       IOC-UPC
       Av. Diagonal 647, planta 11
       08028 Barcelona, Spain

       tel. + 34 93 401 6548
       fax. + 34 93 401 6605
       mailto:raul.suarez@upc.edu
       URL: http://iocnet.upc.edu/usuaris/RaulSuarez

-- 

Jordi Pages, PhD

Head of Intra-logistics Solutions & TIAGo Product Manager

<http://pal-robotics.com/event/euroshop/>

*You can also find us in:*

*European Robotics Forum (3-5 **March in **Malaga, Spain)*
<http://pal-robotics.com/event/european-robotics-forum-2020/>

*Advanced Factories (3-5 March in Barcelona, Spain)
<http://pal-robotics.com/event/advanced-factories/>*

C/ Pujades 77-79, 4-4, 08005 Barcelona, Spain

Skype jordi.pal-robotics

Tel +34 93 414 53 47

jordi.pages@pal-robotics.com

www.pal-robotics.com
Facebook <http://www.facebook.com/palrobotics1> | Twitter
<http://twitter.com/#%21/palrobotics> | YouTube
<http://www.youtube.com/user/PALRobotics> | Blog
<http://blog.pal-robotics.com/>
P Antes de imprimir este e-mail piense bien si es necesario hacerlo: El
medioambiente es cosa de todos.

AVISO DE CONFIDENCIALIDAD: Este mensaje y sus documentos adjuntos, pueden
contener información privilegiada y/o confidencial que está dirigida
exclusivamente a su destinatario.  Si usted recibe este mensaje y no es el
destinatario indicado, o el empleado encargado de su entrega a dicha
persona, por favor, notifíquelo inmediatamente y remita el mensaje original
a la dirección de correo electrónico indicada. Cualquier copia, uso o
distribución no autorizados de esta comunicación queda estrictamente
prohibida.
CONFIDENTIALITY NOTICE: This e-mail and the accompanying document(s) may
contain confidential information which is privileged and intended only for
the individual or entity to whom they are addressed.  If you are not the
intended recipient, you are hereby notified that any disclosure, copying,
distribution or use of this e-mail and/or accompanying document(s) is
strictly prohibited.  If you have received this e-mail in error, please
immediately notify the sender at the above e-mail address.