pal-robotics/tiago_robot #3: urdf joint config doesn't make sense

huiwenzhang, 2018-06-07:
Hi, I am trying to get the `gripper_grasp_frame` pose in gazebo. However, gazebo will automatically merge fixed links to be a whole object, which makes gripper grasp link not accessible in gazebo model. The pose I can get is the `arm_7_link`, so I tried to get the grasp link pose by manually transform the pose from arm 7 link to grasp link. To do this, I need to get the D-H parameter for this transformation. I looked into your urdf file and found the following setting for one of the joints:
https://github.com/pal-robotics/tiago_robot/blob/2892fb78470a139cc22de5d447677616645709a1/tiago_description/urdf/gripper/gripper.urdf.xacro#L123
However, I don't think the setting showed above is correct, given that the relative pose the gripper_link and the grasp link is visualized as follows:
/home/huiwen/Workspace/rviz_screenshot_2018_06_07-03_36_01.png
Is it a mistake or I miss something there? By the way, is there any easier way to get information I need? 
Thanks!

awesomebytes, 2018-06-07:
https://github.com/pal-robotics/tiago_robot/blob/2892fb78470a139cc22de5d447677616645709a1/tiago_description/robots/tiago_steel.urdf.xacro#L38-L41

The parent of the gripper is arm_tool_link:

https://github.com/pal-robotics/tiago_robot/blob/2892fb78470a139cc22de5d447677616645709a1/tiago_description/urdf/arm/arm.urdf.xacro#L261-L265

Which is arm_7_link with a rotation applied and a bit of Z displacement. The rotation effectively makes the X axis look "going out" of the arm. So if you apply that 0.12 on X, well, that's where the grasping frame is. Checking the Rviz TF tree is maybe easier to see the actual tree and it's transformations. In the TF widget you can check the transforms in relation to the previous frames and so.

Good luck.

huiwenzhang, 2018-06-07:
@awesomebytes  Hi, thanks for your replay. Maybe I didn't make it clear. the joint setting from `arm_7_link` to `arm_tool_link` is fine, as you posted above. Where I am confused is the joint setting between `gripper_link` and `gripper_grasp_frame`. As the picture shows, it should be `xyz=0 0 -0.12` or something.

awesomebytes, 2018-06-07:
So the `gripper_grasping_frame` has as parent `gripper_link`:

https://github.com/pal-robotics/tiago_robot/blob/2892fb78470a139cc22de5d447677616645709a1/tiago_description/urdf/gripper/gripper.urdf.xacro#L120-L124

`gripper_link` has as parent `arm_tool_link` and has an applied the `origin` block:

https://github.com/pal-robotics/tiago_robot/blob/2892fb78470a139cc22de5d447677616645709a1/tiago_description/urdf/gripper/gripper.urdf.xacro#L46-L51

Checking the main URDF:
https://github.com/pal-robotics/tiago_robot/blob/2892fb78470a139cc22de5d447677616645709a1/tiago_description/robots/tiago_steel.urdf.xacro#L39-L41

Yeah, it makes no sense.

Dumping the URDF from the parameter server (after it has been passed thru xacro):
```xml
  <!-- Grasping frame -->
  <joint name="gripper_grasping_frame_joint" type="fixed">
    <parent link="gripper_link"/>
    <child link="gripper_grasping_frame"/>
    <origin rpy="-1.5708 1.5708 0" xyz="0 0 -0.12"/>
  </joint>
```
Looks like the rotation has been computed internally... and the XYZ has changed on it.

I double checked using xacro manually:
```bash
roscd tiago_description/robots
rosrun xacro xacro tiago_steel.urdf.xacro > steel.urdf
```
And I get the same result.

Also Rviz, obviously, shows it as the generated URDF.

[image]

Hmm I can only guess it's something we are missing on interpreting how the URDF/xacro works, or a bug in URDF/xacro @rhaschke do you have any idea about this? (I'm voting on us not understanding something, tho).

awesomebytes, 2018-06-07:
This is quite amusing.

```python
from tf.transformations import quaternion_from_euler, euler_from_quaternion
import math
quat = quaternion_from_euler(math.radians(90), math.radians(90), math.radians(-90))
# array([  7.07106781e-01,   5.55111512e-17,  -7.07106781e-01, 1.66533454e-16])
roll, pitch, yaw = euler_from_quaternion(quat)
# (3.141592653589793, 1.5707963267948966, 0.0)
```
That rotation doesn't even match with the one we are getting. Going one step back, (from the screenshot) the rotation of `gripper_link` to `arm_tool_link` is:
```python
euler_from_quaternion([0.70711, 1.7312e-12, -0.70711, 5.1936e-12])
# (1.5707963267961207, 1.57079632679, -1.5707963267948966)
```
Confirmed in the generated URDF too:
```xml
  <joint name="gripper_joint" type="fixed">
    <origin rpy="1.57079632679 1.57079632679 -1.57079632679" xyz="0.01 0 0"/>
    <parent link="arm_tool_link"/>
    <child link="gripper_link"/>
    <axis xyz="0 0 0"/>
  </joint>
```
Which, hey, is what we expected, 90deg, 90deg, -90deg.

Who is rotating and translating the `gripper_grasping_frame`?

awesomebytes, 2018-06-07:
Found it, we were looking in the wrong place, it's not using that gripper URDF, it's using this one:
https://github.com/pal-robotics/pal_gripper/blob/5ee011ffa60612da35432998eb72fa97e3e10071/pal_gripper_description/urdf/gripper.urdf.xacro#L145-L150

from the `pal_gripper_description`.

huiwenzhang, 2018-06-07:
Yeah, you are right. I just overlooked this detail. It spawns the robot by launch gripper urdf from pal package, which is declared here:
https://github.com/pal-robotics/tiago_robot/blob/2892fb78470a139cc22de5d447677616645709a1/tiago_description/robots/tiago_steel.urdf.xacro#L26-L27
Thanks for your  feedback and sorry for the trouble brought to you.

closed 2018-06-07 by huiwenzhang

awesomebytes, 2018-06-07:
@huiwenzhang Best of luck!

bmagyar, 2018-06-07:
You could submit a PR suggesting to remove the superfluous urdf