pal-robotics/tiago_robot #5: Missing rosdep definition in installation of Tiago Simulation

ZdTmasar, 2019-01-22:
Hi,
I once had a working setup of Ros with Tiago working on it but recently, I had to reinstall it and I keep having the same issue over and over even reinstalling everything.

On the tutorial : http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/TiagoSimulation
At the line : rosdep install --from-paths src --ignore-src --rosdistro kinetic --skip-keys="opencv2 opencv2-nonfree pal_laser_filters speed_limit sensor_to_cloud hokuyo_node libdw-dev python-graphitesend-pip python-statsd pal_filters pal_vo_server pal_usb_utils", I keep getting the following message :

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
pmb2_rgbd_sensors: Cannot locate rosdep definition for [pal_pcl]
pmb2_2dnav: Cannot locate rosdep definition for [pal_karto]

I tried to compile but it also fails. Do you have an idea from where ithe problem might come from ?

ZdTmasar, 2019-01-22:
Thank you for your fast answer.

I don't understant why but now it seems to compilate (with warnings and 10 packages skipped).
But the command : roslaunch tiago_gazebo tiago_gazebo.launch public_sim:=true robot:=titanium doesn't work.

Here is the errors :

... logging to /home/thomas/.ros/log/34cdfaa8-1e92-11e9-9c39-7c5cf825152f/roslaunch-thomas-G551JX-9292.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://ros.org/wiki/xacro (/home/thomas/tiago_public_ws/src/pmb2_robot/pmb2_description/urdf/base/base_sensors.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://ros.org/wiki/xacro (/home/thomas/tiago_public_ws/src/tiago_robot/tiago_description/urdf/torso/torso.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://ros.org/wiki/xacro (/home/thomas/tiago_public_ws/src/tiago_robot/tiago_description/urdf/arm/arm.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://ros.org/wiki/xacro (/home/thomas/tiago_public_ws/src/tiago_robot/tiago_description/urdf/head/head.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://ros.org/wiki/xacro (/home/thomas/tiago_public_ws/src/pal_gripper/pal_gripper_description/urdf/gripper.urdf.xacro)
deprecated: xacro tags should be prepended with 'xacro' xml namespace.
Use the following script to fix incorrect usage:
        find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
when processing file: /home/thomas/tiago_public_ws/src/tiago_robot/tiago_description/robots/tiago_steel.urdf.xacro

Invalid parameter "sick561" None None
when instantiating macro: base_sensors (/home/thomas/tiago_public_ws/src/tiago_robot/tiago_description/robots/tiago_steel.urdf.xacro)
in file: /home/thomas/tiago_public_ws/src/tiago_robot/tiago_description/robots/tiago_steel.urdf.xacro
while processing /home/thomas/tiago_public_ws/src/tiago_simulation/tiago_gazebo/launch/tiago_spawn.launch:
while processing /home/thomas/tiago_public_ws/src/tiago_robot/tiago_description/robots/upload_tiago.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/thomas/tiago_public_ws/src/tiago_robot/tiago_description/robots/tiago_steel.urdf.xacro' ] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro.py '$(find tiago_description)/robots/tiago_$(arg robot).urdf.xacro' $(arg multi)" name="robot_description"/>
The traceback for the exception was written to the log file

v-lopez, 2019-01-23:
We have updated all our tutorials and tiago packages.

I recommend you delete your downloaded workspace and start the installation tutorial again, since we've changed some steps.

Let me know if you encounter any issue.

ZdTmasar, 2019-01-23:
Everything is working fine at first sight. I launched the simulation at the end of the installation and the pick&place simulation and I didn't notice anything wrong.
Thank you for your help :)

closed 2019-01-23 by ZdTmasar