pal-robotics/tiago_robot #6: install DESTINATION /

olivier-stasse, 2019-04-03:
The following line 
https://github.com/pal-robotics/tiago_robot/blob/aab9e850a63b157122d1dbebddb6763e62ab0bf9/tiago_bringup/CMakeLists.txt#L15
is making a catkin_make install failing because it tries to access the root of the system.

In our packaging system this is also failing.

It is probably making sense on the real robot but for simulation, CI or local deployment this seems to be a problem.

Would it be ok to add a cmake variable to disable this install. For you to have no change it would be set to a default value performing the install.

v-lopez, 2019-04-03:
Hi @olivier-stasse, thanks for the fix suggestion.

We'll be moving this rule to a separate package, because it really doesn't belong here.

I'm pushing an update that fixes this: https://github.com/pal-robotics/tiago_robot/commit/702f87c3c0c42d018252584a96dcd8c50a8cd537

olivier-stasse, 2019-04-03:
Fixed by 702f87c

olivier-stasse, 2019-04-03:
Thanks for the release !

closed 2019-04-03 by olivier-stasse