pal-robotics/tiago_robot #11: Wrong position of the torso first collision link

jmirabel, 2021-02-10:
It seems that the origin of the first collision box of the torso is not correct. I expect it to be
```xml
 <origin rpy="0 0 0" xyz="0 0 0.25" /> 
```

https://github.com/pal-robotics/tiago_robot/blob/31790ca2cd84b4f591ff22b497127e0129bfc4ca/tiago_description/urdf/torso/torso.urdf.xacro#L46

This is related to https://github.com/pal-robotics/pmb2_robot/issues/6

v-lopez, 2021-02-10:
Could you please elaborate on this?

Visually in rviz and gazebo I see them align, and if I increase z by 0.20 cm it is floating in the middle of the torso.

They are aligned by default:
[image]

jmirabel, 2021-02-10:
Actually, you are right, the issue may not be that the box is not well placed indeed. 

When I measure with a ruler on the robot the height of the small step where the arm is placed when off, I get something between 45 and 50mm. If I do the same in rviz, with collision enabled, along the black line below, I get about 33mm.
[image]

v-lopez, 2021-02-10:
Yeah that could be, again we'll pass this to the mechanical department so they can generate more up to date collision meshes.

Thanks for reporting it.