pal-robotics/tiago_robot #14: Missing pal_local_joint_control

andreagavazzi, 2021-08-17:
tiago_controller_configuration package, has dependency on "pal_local_joint_control" that seems not available anywhere

saikishor, 2021-08-18:
@andreagavazzi Thanks for showing interest in our TIAGo. Regarding your query, `pal_local_joint_control` is our internal package for the joint level control. However, this dependency is not needed for you to run the TIAGo public simulation.

You can find the info of how to setup and run the simulation here: http://wiki.ros.org/Robots/TIAGo/Tutorials

Thank you,

Best Regards,
PAL Robotics Team

saikishor, 2022-02-11:
Hello @andreagavazzi,

We will be closing this issue due to inactivity.

Best Regards,
PAL Robotics Team

closed 2022-02-11 by saikishor