pal-robotics/tiago_robot #15: Self-Collision problem with Tiago URDF

i-m-vivek, 2021-09-22:
[image]

While following the Tiago [Moving Individual Joints](http://wiki.ros.org/Robots/TIAGo/Tutorials/motions/rqt_joint) tutorial I found the robot hand goes inside the robot. I found this problem while using the Tiago URDFs in PyBullet.  

**To reproduce:** 
- Follow the same tutorial.
- Set the arm joints values as shown in the right figure.

saikishor, 2021-09-22:
@i-m-vivek Thanks for showing interest in our TIAGo Simulation.

Regarding your query of the joints ending up with self-collision. The joint trajectory controllers doesn't have any idea of the robots collision meshes and it doesn't take care of it. Its sole purpose is to move the joints within one joint limit to another joint limit, even if it ends up in self collision.

If you are interested to avoid cases of self-collision. You should be looking into moving the joints with MoveIt. The self collision configuration is encoded within the package tiago_moveit_config package in form of .srdf. 

Please feel free to get back to us, in case you have more queries.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-02-11:
Hello @vk-mittal14,

We will be closing this issue due to inactivity. We hope you found the solution.

Have a great day,

Best Regards,
PAL Robotics Team

closed 2022-02-11 by saikishor