pal-robotics/tiago_robot #16: Starting a Real TiaGo Robot

linhdoan8, 2022-02-06:
Dear Pal Team, 

first of all, thanks for the great job! 

While the gazebo simulation is working fine, there are still some problems when working with the real TiaGo robot.
My real TiaGo is not working properly and I have installed Ubuntu 16 and ROS kinetic on its on-board PC. My goal is to start the TiaGo bot and all its sensors and actuators for navigation. However, when running    
```
roslaunch tiago_bringup tiago_bringup.launch 
```
this error appears: 
```
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 207, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: pal_local_joint_control
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/pal/code/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
```

I cannot find the package pal_local_joint_control anywhere and in one of your answers, you have mentioned that it is an internal package of yours. Will it be released anytime soon? What other alternative is there to start the real TiaGo motors and sensors? Is there any documentation online to setup and start a real TiaGo robot?

In case you ask: the natively deployed OS (with Ubuntu 14.04.) is not working properly, it doesnt broadcast a network to connect to and also no laserscan and odom is published. The laser scan launch files seem corrupted when trying to start them manually. Teleoperation is working fine though and the robots head and arms are moving, the lights on the side are on and I can teleoperate the robot. However, everything else seem broken, which is why I decided to install Ubuntu from scratch and run the packages manually.

Thank you!

saikishor, 2022-02-06:
Dear @linhdoan8,

Thank you for writing to us. Regarding your issue, you cannot install the standard Ubuntu and ROS on a normal development machine. The robot usually comes with Ubuntu, standard ROS packages, and an overlayed ROS distribution from PAL Robotics, this also includes the RT Kernel patch, motor configurations, startup manager, and so on. You won't be able to run the TIAGo Robot without this configuration and package setup.

In order to have the issue properly resolved, you will need to install the distribution provided by PAL Robotics. You can open a support ticket at [support.pal-robotics.com](https://support.pal-robotics.com/) and request support for this. We hope this helps to solve your issue.

Thank you,

Best Regards,
PAL Robotics Team.

linhdoan8, 2022-02-06:
Dear @saikishor ,

thank you very much for your swift reply! Good to know! However, I am not able to login to the support portal as I am not a member. Also, the PAL websites are all down.

I have found two USB sticks, one with "Tiago Software", one with "Tiago System". The "Tiago System" Stick will install a very basic Ubuntu 14 terminal-only version. I assume that the second USB stick with "Tiago Software" will install the ROS distribution from PAL. If it is possible here, can you tell me, how I can install it with this stick? Or if there is any documentation online? I tried booting with the stick but it doesn't seem to work. Also, I don't see any option to install it from the terminal-only Ubuntu 14 distro installed with the other stick. Unfortunately, there is no printed documentation or anything related to the TiaGo robot left in the lab besides those 2 Sticks. Hope for your reply.

Thank you very much and best regards,
Linh

saikishor, 2022-02-06:
Hello @linhdoan8,

Thank you for getting back to us. The ISOs we provide usually contain 2 installations, one for the robot and another for the development computer. We apologize for the issues you are facing with the website and the support system. I kindly request you to drop in an email to the <support@pal-robotics.com> along with your specific TIAGo serial number and details, so we can check tomorrow and get back to you ASAP.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-02-09:
Hello @linhdoan8,

I'll be closing this ticket and you have already been in contact with our support team. We wish you a great day.

Thank you,

Best Regards,
PAL Robotics Team

closed 2022-02-09 by saikishor

linhdoan8, 2022-02-11:
Dear @saikishor,

thanks for your help! After contacting the support team, they sent me documentation on how to install the Tiago system from scratch using the 2 ISOs. In case someone faces the same experience, I have described the steps [here](https://github.com/ignc-research/real-robots).

Thanks again and best regards,
Linh

saikishor, 2022-02-11:
Hello @linhdoan8,

We are happy to know that your issue is resolved. We hope you will enjoy your experience with our TIAGo. 

Have a great day,

Best Regards,
PAL Robotics Team.