pal-robotics/tiago_robot #18: pmb2_description.pmb2_launch_utils no longer exists

fmrico, 2022-09-19:
Dear PAL team,

This code seems not to be longer valid:

https://github.com/pal-robotics/tiago_robot/blob/8bf78ce642a5b62bab7ccfc5b4883d72693b4ba3/tiago_description/tiago_description/tiago_launch_utils.py#L16

Any idea to fix it?

Thanks!!

saikishor, 2022-09-19:
Hello @fmrico,

Thank you for writing to us. Regarding the tiago_launch_utils.py the method still exists in the following file:

https://github.com/pal-robotics/pmb2_robot/blob/58cba9be1a5ca8b3a043d93924da252ab6f8991f/pmb2_description/pmb2_description/pmb2_launch_utils.py#L18

We hope this helps to resolve your issue,

Best Regards,
PAL Robotics Team.

fmrico, 2022-09-19:
Hi @saikishor 

I have just solved it by using https://github.com/pal-robotics/pmb2_robot/tree/07e0fcd82cc0d21e0a03cb020bd6d69d5be5574d

Thanks anyway!!
Francisco

closed 2022-09-19 by fmrico