pal-robotics/tiago_robot #20: ros2_control: Param file does not contain parameters for joint_state_broadcaster

AndyZe, 2023-03-30:
I'm trying to launch Tiago with the Humble debians, but I see errors like this for every controller:

```
tiago_behavior_demos-sim-1  | [spawner-9] [INFO] [1680183611.891801604] [spawner_arm_controller]: Set controller type to "joint_trajectory_controller/JointTrajectoryController" for arm_controller
tiago_behavior_demos-sim-1  | [gzserver-1] [INFO] [1680183611.893191959] [controller_manager]: Loading controller 'arm_controller'
tiago_behavior_demos-sim-1  | [gzserver-1] [INFO] [1680183611.920356180] [controller_manager]: Setting use_sim_time=True for arm_controller to match controller manager (see ros2_control#325 for details)
tiago_behavior_demos-sim-1  | [spawner-9] [INFO] [1680183611.944682539] [spawner_arm_controller]: Loaded arm_controller
tiago_behavior_demos-sim-1  | [spawner-9] Traceback (most recent call last):
tiago_behavior_demos-sim-1  | [spawner-9]   File "/opt/ros/humble/lib/controller_manager/spawner", line 33, in <module>
tiago_behavior_demos-sim-1  | [spawner-9]     sys.exit(load_entry_point('controller-manager==2.24.1', 'console_scripts', 'spawner')())
tiago_behavior_demos-sim-1  | [spawner-9]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 212, in main
tiago_behavior_demos-sim-1  | [spawner-9]     load_parameter_file(node=node, node_name=prefixed_controller_name, parameter_file=param_file,
tiago_behavior_demos-sim-1  | [spawner-9]   File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/api/__init__.py", line 149, in load_parameter_file
tiago_behavior_demos-sim-1  | [spawner-9]     raise RuntimeError('Param file does not contain parameters for {}, '
tiago_behavior_demos-sim-1  | [spawner-9] RuntimeError: Param file does not contain parameters for arm_controller,  only for nodes: dict_keys(['/arm_controller'])
```

I think the config file it refers to is here:  https://github.com/pal-robotics/tiago_robot/blob/humble-devel/tiago_controller_configuration/config/head_controller.yaml

Noel215, 2023-03-30:
Hi @AndyZe,

That issue is already solved and released, it will be fixed with the next humble sync (tomorrow).
You can wait for that or clone the repositories [tiago_robot](https://github.com/pal-robotics/tiago_robot/tree/humble-devel) and [pal_gripper](https://github.com/pal-robotics/pal_gripper/tree/humble-devel) in your workspace.

AndyZe, 2023-03-30:
Thank you for the fast response! That's great. Out of curiosity, was this the commit that fixed it?

https://github.com/pal-robotics/tiago_robot/commit/041dc7552f9d54e14c0528a726f83276cd827a5b

I'll close this issue after verifying that it works tomorrow   :+1:

Noel215, 2023-03-30:
You are right! That was the commit that fixed the issue

AndyZe, 2023-04-01:
It works!

closed 2023-04-01 by AndyZe