pal-robotics/tiago_robot #22: How to create urdf from xacro - missing meshes

adamkrawczyk, 2023-06-07:
Hi, 

SPEC:
- Ubuntu 22
- ROS2 humble

I have started exploring the project but am stuck on converting xacro to urdf (I want to use it in a simulator other than gazebo). 

1. I have cloned and built the repository (assume that `colcon build --symlink-install`)
2. Try to convert xacro (`xacro robots/tiago.urdf.xacro > tiago.urdf`)

There are missing meshes in the whole `package://pmb2_description/meshes/base/base.stl` I have no directory `base` under meshes.
Can I generate these files or how should I get them? 

Best, 
Adam Krawczyk

adamkrawczyk, 2023-06-07:
@saikishor 

Hmmm strange, could you provide a link? 

https://github.com/pal-robotics/tiago_robot/tree/humble-devel/tiago_description/meshes 

I have searched but there is nothing.

saikishor, 2023-06-07:
Hello @adamkrawczyk,

I just provided you with the link above in my previous reply. Please take a look.

Best Regards,
PAL Robotics Team.

adamkrawczyk, 2023-06-13:
@saikishor 
Thank you and I apologize for the issue. I have installed the required packages from apt (for build). 

It seems that everything is resolved now. (The same thing with `hey5_description`).

closed 2023-06-13 by adamkrawczyk

kavikode, 2024-08-24:
How do we create the final URDF with the default values from the tiago_description/config/tiago_configuration.yaml? @adamkrawczyk 

How do we create the final URDF with the hey5_description end effector? @saikishor 

thank you

saikishor, 2024-09-02:
> How do we create the final URDF with the default values from the tiago_description/config/tiago_configuration.yaml? @adamkrawczyk
> 
> How do we create the final URDF with the hey5_description end effector? @saikishor
> 
> thank you

@kavikode Why don't you just launch the simulation or the description and just save the published URDF in `/robot_description` on to a file

adamkrawczyk, 2024-09-02:
Hi, @saikishor 

I have used this to create a model for my O3DE sim here is link to the repo: [link](https://github.com/adamkrawczyk/TiagoSimulation)
hmmm, I see that I didn't provided the exact packages I have used and I have it on the other PC (will share to you later). So far here is the diagram of the used packages (sorry but don't have time now) 
[image]

Just take the packages listed for the tiago sim (to your ros2_ws) build and if I remember correctly you will be able to generate urdf from the xacro inside `tiago_robot`

adamkrawczyk, 2024-09-02:
@saikishor btw: If you want, you can also use this o3de sim - when last tested it was working

saikishor, 2024-09-02:
Hello @adamkrawczyk!

Thank you for sharing your resources with O3DE. @kavikode just wants to export the URDF into a single file instead of the modular filesystem we have. That's what I'm suggesting here to use it from the published topic in case of ROS 2 and the parameter in the case of ROS

adamkrawczyk, 2024-09-02:
@saikishor Then it's doable (I did that pretty easy - just build the ros2_ws with the packages) and if not from the o3de then with open-sourced tools you can do it (don't remember which one i have used but did that - you have to google).

kavikode, 2024-09-05:
I appreciate your response.

I tried the following:

Terminal 1: 
ros2 launch tiago_gazebo tiago_gazebo.launch.py moveit:=True world_name:=pick is_public_sim:=True end_effector:=pal-hey5

Terminal 2:
ros2 topic echo /robot_description > tiago_pal_hey5.urdf

I'm still just getting a blank file as follows:
data: <?xml version="1.0" ?>

saikishor, 2024-09-05:
@kavikode you can expand the contents by passing option '-f' to ros2 topic echo. Try checking more options with -h. I hope this helps