pal-robotics/tiago_robot #23: Tiago robot self-rotate in gazebo

nicehiro, 2023-06-26:
I'm encountering an issue recently that the tiago robot rotate by itself slightly in gazebo simulation.

Reproducing process:
- start simulation, and rotate tiago by a certain degree, and then stop tiago movement

Then tiago start self-rotating backwards slightly.

There's some same issues can be found here (one answer are using tiago too):
- https://answers.ros.org/question/310008/turtlebot-and-other-objects-unintentionally-moving-by-itself-in-gazebo/
- https://answers.ros.org/question/313899/banana-rotates-by-itself-in-gazebo-i-dont-know-why/
- https://answers.gazebosim.org/question/21944/banana-rotates-by-itself-in-gazebo-i-dont-know-why/

I've tried all the answers above, but still couldn't solve the issue.

saikishor, 2023-06-26:
Hello @nicehiro,

We will try to take a look at our end and get back to you!

Thank you,

Best Regards,
PAL Robotics Team.

nicehiro, 2023-06-28:
Hi, @saikishor,

Sorry to bother you. Did you find any solutions?

I tried to open a ticket in PAL support center but it seems that the website is temporarily unavailable.

I also send emails to PAL but got no response.

thomaspeyrucain, 2023-06-28:
Hello @nicehiro,

Thank you for contacting us,

I could not reproduce your issue 
How are you launching the rotation process? Through navigation or key_teleop?
Which gazebo and ROS versions are you using?
How slight is the movement? Could you maybe share a video

Best regards,
PAL Robotics Team

closed 2023-06-28 by saikishor