pal-robotics/tiago_robot #24: Error loading config file with no-arm selection

jmguerreroh, 2023-06-28:
Running tiago_gazebo, with no-arm parameter:
```
ros2 launch tiago_gazebo tiago_gazebo.launch.py arm:=no-arm
```
The path is not correctly generated:
```
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: ~/ros2_ws/install/tiago_bringup/share/tiago_bringup/config/motions/tiago_motions_no-arm_pal-gripper_schunk-ft.yaml
```
The files provided are:

- tiago_motions_no-arm.yaml                    
- tiago_motions_no-end-effector_schunk-ft.yaml  
- tiago_motions_no-end-effector.yaml            
- tiago_motions_pal-gripper_schunk-ft.yaml      
- tiago_motions_pal-gripper.yaml
- tiago_motions_pal-hey5_schunk-ft.yaml
- tiago_motions_pal-hey5.yaml
- tiago_motions.yaml.em

Fixing the error as (https://github.com/pal-robotics/tiago_robot/pull/21):
```
if (arm == 'no-arm'):
        motions_file = 'tiago_motions_no-arm.yaml'
```
Then another error appears:
```
[play_motion2_node-11] [ERROR] [1687948338.550539793] [play_motion2]: Service /controller_manager/list_controllers not available.
[play_motion2_node-11] [ERROR] [1687948338.550716131] [play_motion2]: There are no active JointTrajectory controllers available

[play_motion2_node-11] [ERROR] [1687948356.631965118] [play_motion2]: Joint 'arm_1_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687948356.632317282] [play_motion2]: Joint 'arm_2_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687948356.632548396] [play_motion2]: Joint 'arm_3_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687948356.632615411] [play_motion2]: Joint 'arm_4_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687948356.632656069] [play_motion2]: Joint 'arm_5_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687948356.632693787] [play_motion2]: Joint 'arm_6_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687948356.632718715] [play_motion2]: Joint 'arm_7_joint' is not claimed by any active controller
```

Thank you! :)

Noel215, 2023-06-28:
Hi @jmguerreroh,

My bad, should be working now :wink: 
You can see the changes in https://github.com/pal-robotics/tiago_robot/commit/0613fec23cd1f0f20f91903d70b2b01be6d8af29

jmguerreroh, 2023-06-28:
Hi @Noel215 

Thank you for the fix, but after fixing it, I obtain the same problem continuously:

```
[tuck_arm.py-15] [ERROR] [1687959842.282169718] [arm_tucker]: play_motion2 is not ready
[play_motion2_node-11] [ERROR] [1687959845.286888166] [play_motion2]: Joint 'arm_1_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687959845.286967941] [play_motion2]: Joint 'arm_2_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687959845.287026510] [play_motion2]: Joint 'arm_3_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687959845.287072087] [play_motion2]: Joint 'arm_4_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687959845.287106891] [play_motion2]: Joint 'arm_5_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687959845.287142209] [play_motion2]: Joint 'arm_6_joint' is not claimed by any active controller
[play_motion2_node-11] [ERROR] [1687959845.287176907] [play_motion2]: Joint 'arm_7_joint' is not claimed by any active controller
```

Any idea?

Thank you

Noel215, 2023-06-29:
The issue is that the `tuck_arm` node is always run, even if the arm does not exist. I'll patch it

jmguerreroh, 2023-06-29:
> The issue is that the `arm_tucker` node is always run, even if the arm does not exist. I'll patch it

I will wait for the solution. Thank you!

Noel215, 2023-07-05:
Hi @jmguerreroh, 

The issue was related to the config generator, I've already fixed it.

Thanks for reporting it.

jmguerreroh, 2023-07-05:
> Hi @jmguerreroh,
> 
> The issue was related to the config generator, I've already fixed it.
> 
> Thanks for reporting it.

Hi @Noel215 

It works like a charm, thank you!

closed 2023-07-05 by jmguerreroh