pal-robotics/tiago_robot #25: Missing pal_motion_loader package in simulation (ROS Noetic, kinetic-devel branch)

PeterBowman, 2023-11-29:
After following the [installation guide for ROS Noetic](http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/InstallUbuntuAndROS), we tried to launch the robot in simulation via:

 ```
roslaunch tiago_gazebo tiago_gazebo.launch public_sim:=true end_effector:=pal-gripper
```

It throws a runtime error related to a missing `pal_motion_loader` package (simulation starts, but the planner doesn't, and the tuck_arm node gets stuck). It does work on a previous installation on the sources we downloaded back in September 2023. To sum up:

- Simulation works at old version 2.3.13 (after applying https://github.com/pal-robotics/tiago_robot/commit/238e7eaf32513d9a134986950972b0d1524aeab4).
- Simulation breaks at current version 2.3.17.
- Cause: now the planner requires an additional `pal_motion_loader` that is **not** installed alongside other repos via rosinstall ([diff](https://github.com/pal-robotics/tiago_robot/compare/890829a71782e4c3c5444fae670bdc04b5fde057...kinetic-devel)).

On a side note, during the `catkin build` phase, there are a few CMake errors caused by `find_package(REQUIRED)`. It is necessary to install them manually via `apt install ros-noetic-<package>`. I presume the rosinstall script is incomplete.

PeterBowman, 2023-11-29:
Besides, the following instruction listed at the end of the [installation guide](http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/InstallUbuntuAndROS) also throws a dependency error due to not being able to find a `pal_motion_loader` package:

```
rosdep install -y --from-paths src --ignore-src --rosdistro noetic --skip-keys "urdf_test omni_drive_controller orocos_kdl pal_filters libgazebo9-dev pal_usb_utils speed_limit_node camera_calibration_files pal_moveit_plugins pal_startup_msgs pal_local_joint_control pal_pcl_points_throttle_and_filter current_limit_controller hokuyo_node dynamixel_cpp pal_moveit_capabilities pal_pcl dynamic_footprint gravity_compensation_controller pal-orbbec-openni2 pal_loc_measure pal_map_manager ydlidar_ros_driver"
```

saikishor, 2023-11-29:
Hello @PeterBowman,

You are right. There is a missing package to be public. We are trying to solve the issue at our end. Once it is solved, we will get back to you.

Thank you,

Best Regards,
PAL Robotics Team.

thomaspeyrucain, 2023-11-29:
Hello @PeterBowman,

The package is now available on github. if you redownload (wget https://raw.githubusercontent.com/pal-robotics/tiago_tutorials/noetic-devel/tiago_public-noetic.rosinstall) the rosinstall and redo the steps it should work now

The needed package is available here if needed: [https://github.com/pal-robotics/pal_motion_loader](https://github.com/pal-robotics/pal_motion_loader)

Best Regards,
PAL Robotics Team

closed 2023-11-29 by saikishor