pal-robotics/tiago_robot #27: Error:  /move_base    Range sensor layer can't transform from map to base_sonar_01

gianluca-cioccolini, 2024-04-18:
We are encountering a problem with our robot Tiago. We have the following errors:

```
10:18:10:364    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176290.212631
10:18:09:168    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176289.011473
10:18:08:064    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176287.879702
10:18:06:630    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176286.447712
10:18:05:564    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176285.431382
10:18:04:365    /move_base    Range sensor layer can't transform from odom to base_sonar_01_link at 1713176284.219819
10:18:03:364    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176283.218585
10:18:02:165    /move_base    Range sensor layer can't transform from odom to base_sonar_01_link at 1713176282.011889
10:18:01:165    /move_base    Range sensor layer can't transform from odom to base_sonar_01_link at 1713176281.010118
10:18:00:165    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176280.008849
10:17:59:099    /move_base    Range sensor layer can't transform from odom to base_sonar_01_link at 1713176278.902425
10:17:56:164    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176276.010399
10:17:55:163    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176275.009291
10:17:54:163    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176274.008252
10:17:52:965    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176272.805622
10:17:51:965    /move_base    Range sensor layer can't transform from map to base_sonar_01_link at 1713176271.804663
```
Is this error affecting the functionality of Tiago?  What steps can be taken to resolve this issue?
Any assistance or guidance on resolving this error would be greatly appreciated.

Shikigami1606, 2025-02-28:
Your Tiago robot is complaining that it can't transform data from map or odom to base_sonar_01_link. This basically means that the range sensor (probably a sonar or similar sensor) isn't getting the correct positional info from the robot’s navigation system. That can mess with obstacle detection and navigation.

This might be happening because -

   1. The transform (TF) linking these frames might not be set up properly.
   2. There could be a delay or missing transform being published.
   3. The sonar frame (base_sonar_01_link) might not exist in the TF tree, or something’s off in the configuration.
   4. If the transform is being broadcasted, maybe there's a timing issue or a missing static transform.

onemriganka, 2025-03-25:
/attempted #27