pal-robotics/tiago_robot #30: Invalid inertia values

EGAlberts, 2024-08-05:
In the process of migrating from Gazebo classic to Gazebo fortress, I am publishing the robot_description described contained within this package. In doing so, I get three errors:
```
[Err] [UserCommands.cc:1138] Error Code 18: Msg: A link named arm_1_link has invalid inertia.
[Err] [UserCommands.cc:1138] Error Code 18: Msg: A link named gripper_left_finger_link has invalid inertia.
[Err] [UserCommands.cc:1138] Error Code 18: Msg: A link named gripper_right_finger_link has invalid inertia.
```

By zero-ing out the inertia values (see https://github.com/EGAlberts/tiago_robot/commit/a4a7fd658271c74c1543be083b49007ffa3ebc40) of these three links I am able to make Tiago spawn.

One of these lives in this repo, specifically arm_1_link. I shall open a separate issue in the pal_gripper repository.

Can you please verify the inertia values are valid? Or determine if the parsing/validation by the new version of Gazebo warrants changes?

EGAlberts, 2024-08-05:
To reproduce this issue you can simply use ros2 launch tiago_description robot_state_publisher.launch.py to publish the robot_description, and this kind of snippet with gazebo fortress running to spawn based on the published robot_description:

```
robot_entity = Node(
    package="ros_gz_sim",
    executable="create",
    arguments=[
        "-topic",
        "robot_description",
        "-name",
        LaunchConfiguration("robot_name"),
    ],
    output="screen",
)
```

saikishor, 2024-08-09:
Hello @EGAlberts!

Thank you for reporting the bug. Most of our team are currently on vacation. We will take a look into it when our team is back.

Best Regards,
PAL Robotics Team

EGAlberts, 2024-08-09:
Thanks for replying, in the mean time I used meshlab and imported the .stl of Tiago's shoulder. This gave me new inertia values which do make sense. https://www.hamzamerzic.info/mesh_cleaner/ is a website which allows you to do this process without actually installing meshlab yourself and just upload the relevant mesh file. See the relevant commit: https://github.com/EGAlberts/tiago_robot/commit/d95a395c3ed4d32c59cc8e5faf045527f5faa616 . You may consider recreating this process yourself to confirm our work, and possibly having those be the inertia values. To give more details about the error with the current values, it finds that the triangle equality (some moments are smaller when summed than bigger ones) is not satisfied.