pal-robotics/tiago_robot #37: adding pal dependencies in README

Aniketkumarroy, 2025-04-11:
Hi @saikishor
the repository depends on some pal dependencies which are not given in README, if someone wants to work on it, building the package throws errors for missing dependencies, one at a time, which is quite annoying. shouldn't we add an instruction on how to include the bare minimum pal dependencies for compiling the package, like either installing binaries or cloning them and building? it will be much more convenient for any newcomer

awesomebytes, 2025-04-11:
The dependencies are in the package.xml files declared as such and rosdep will resolve them. Generally like all ROS packages do

saikishor, 2025-04-11:
Thank you @awesomebytes absolutely on point. I'll go ahead and close the issue as it can be solved with `rosdep` command.

closed 2025-04-11 by saikishor

Aniketkumarroy, 2025-04-12:
@awesomebytes Yes I got your point but what I want to convey is that we can put that information in README just like [tiago_simulation](https://github.com/pal-robotics/tiago_simulation) has an extensive guide.
additionally, we can include a `vcs import` utility to clone the dependencies in the local workspace instead of installing it on system. why?? because many(even I) like to clone the less complex dependencies locally so that we can tweak them and even if we messed something this doesn't hurt at the system level.

it's completely my perspective and maybe I am wrong so apologies.

saikishor, 2025-04-13:
> @awesomebytes Yes I got your point but what I want to convey is that we can put that information in README just like [tiago_simulation](https://github.com/pal-robotics/tiago_simulation) has an [...]

@Aniketkumarroy it is not recommended for various reasons. It is added maintenance for the Maintainers + it's hard to leave one that works for everyone. There are some internal repos too. For this reasons, we won't be able to consider this request